njz3/vJoyIOFeederWithFFB #2: Multi axis force feedback

yuvalsegev, 2020-10-08:
Hi!

First of all, major props for sharing this project, will surely help many people down the road.

Is there a plan to add force feedback to the Y axis? This would be awesome as it would allow using this as a flight simulator joystick.

njz3, 2020-10-08:
Hello,
This feeder was really meant to be used for a Sega racing cabinet, piloting the heavy motors that is found inside this excellent piece of hardware using Sega's proprietary boards and an Arduino.
I ported back some part of my feeder back into the C# sample in the last vJoy SDK for the ffb message pipe, but I did not port the effect processing engine.
Since I do not own myself a flight simulator joystick, neither have such software, I can only barely test with fedit that effects on X/Y axes work, which is the case as far as I tested. So I doubt I will extend any time soon.
If you want to extend it yourself (knowing a little C# and Arduino), maybe it is worth taking ideas and code from my feeder and build your own. I am sure it is definitely doable.

yuvalsegev, 2020-10-08:
Thanks for the quick response.

Did you write the effect processing code? 
I haven't seen anybody else outputting torque values from a vJoy feeder so I guess you did.

If that's not the case, can you refer to the original repo?
I think I'll just write my own feeder then..

Thanks in advance.

njz3, 2020-10-08:
Most of the code was written from scratch and I looked to the adapt-ffb project to understand what was the meaning of some effects and parameters.
Do you plan to use an arduino ?

yuvalsegev, 2020-10-08:
Yep, planning on building a boeing 737 dual yoke system using high torque DC motors
e.g. [https://www.fsc.it/Gallery/products/219914/description%20images/FSC-B737-YOKE-PRO-2017-FRONT-MAIN.jpg](image)

closed 2020-10-08 by njz3