njz3/vJoyIOFeederWithFFB #7: What about Rumble?

baritonomarchetto, 2022-04-01:
Hello, first of all thanks for sharing such a nice project but also for spending your time in documenting it.
I know that most of the racing games this hardware/software combination support do not support rumble (they are FF games) but would it be possibile to add some support for Rumble setups? In other words, would it be possibile to include a routine that "translates" FF messages into rumble, i.e. by triggering a rotating motor only when the car goes out of track or collides with another car and ignores all the other FF effects?
Thanks :)

njz3, 2022-04-01:
I actually added this into the latest version (not public).
It is PWM_SHAKER mode. The Arduino will output a PWM that can be used as a rumble.

baritonomarchetto, 2022-04-03:
Great!

baritonomarchetto, 2022-09-20:
Sorry for calling back this old request of mine: is PWM_SHAKER mode pubblic now?

njz3, 2022-09-20:
It is available in the latest 0.6 binary version, but source will be kept as closed.

baritonomarchetto, 2022-09-21:
(Benjamin, sorry for bothering you. May I ask why source is closed now? Not that I want to modify or take advantage from it, it's way out of my possibilities, and the exe is more than enought, but it's always a bad news when the development of such a nice project is made unavailable)

njz3, 2022-09-21:
Reasons are that some part of the 0.6 version contains advanced filtering algorithms (state estimator) and many corrections to the ffb engine that I do not want to disclose, and also I want to restrain people from reselling my work for commercial arcade activities, by at least enforcing them to get a licence for each install so that I know who use the software in this small community.

baritonomarchetto, 2022-09-21:
Makes sense. Thanks for your answer

closed 2022-09-21 by baritonomarchetto