njz3/vJoyIOFeederWithFFB #9: Question about vJoy FFB registry, FFB Flight stick needs different effect parameter than wheel

cyberluke, 2022-08-21:
Hi, I noticed with your latest vJoy version, there is no need to add registry settings.

So it is stored somewhere in the source code? I need to change it.

For example, the default for Spring FFB is for wheel:
```
@="GUID_Spring"
"Attributes"=hex:40,00,0f,00,04,c8,00,00,65,03,00,00,65,03,00,00,20,00,00,00
```

But flight stick has this:
```
@="Spring"
"Attributes"=hex:07,00,00,00,04,d8,00,00,6d,03,00,00,6d,03,00,00,30,00,00,00
```

Effect GUID for Spring is the same for both:
OEMForceFeedback\Effects\{13541C27-8E33-11D0-9AD0-00A0C9A06E35}

But this difference causes this kind of exception:
```
----------------------
DataSize: 30, CMD: IOCTL_HID_WRITE_REPORT
1A030101FF000000000000FFFF0424360000000000000000000000000000
BlockIdx: 3
Device ID: 1
Packet type: Effect
	EffectType: Spring
	Duration: -1
	TriggerRepeatInterval: 0
	SamplePeriod: 0
	Gain: 0
	TriggerBtn (?): 0
	Polar: True
	Angle: 360

Receive->process delay: 1,674.999ms
Forwarding to 1 device
Creating effect: Spring
Exception when trying to forward:
	Unable to create new effect: Object reference not set to an instance of an object.
	   at FreePIE.Core.Plugins.Dx.Device.CreateEffect(Int32 blockIndex, EffectType type) in G:\Downloads\FFB\FreePIE-Ffb\FreePIE.Core.Plugins\Dx\Device.cs:line 255
   at FreePIE.Core.Plugins.Dx.Device.CreateEffect(Int32 blockIndex, EffectType effectType, Boolean polar, Int32[] dirs, Int32 duration, Int32 gain, Int32 samplePeriod, Int32 startDelay, Int32 triggerButton, Int32 triggerRepeatInterval) in G:\Downloads\FFB\FreePIE-Ffb\FreePIE.Core.Plugins\Dx\Device.cs:line 154
   at FreePIE.Core.Plugins.Dx.Device.SetEffectParams(EffectReportPacket er) in G:\Downloads\FFB\FreePIE-Ffb\FreePIE.Core.Plugins\Dx\Device.cs:line 120
   at FreePIE.Core.Plugins.VJoy.PacketMapper.<>c.<SetupDefaultMap>b__5_0(Device d, EffectReportPacket p) in G:\Downloads\FFB\FreePIE-Ffb\FreePIE.Core.Plugins\VJoy\PacketMapper.cs:line 25
   at FreePIE.Core.Plugins.VJoy.AsyncPacketData`1.Call(IList`1 registeredDevices) in G:\Downloads\FFB\FreePIE-Ffb\FreePIE.Core.Plugins\VJoy\PacketAction.cs:line 65
----------------------
```

I have tested and verified your vJoy successfully with this FreePIE fork with FFB support: https://github.com/cyberluke/FreePIE/tree/Ffb

If I add my own values to the registry, it somehow filters incoming FFB packets and my callback is not called. If I delete the registry, the FFB callback works, but with different parameters it causes this issue. On previous vJoy I experienced this issue and the registry tweak fixed it.

I am now compiling your latest vJoy source...some pointer where to adjust these parameters would be awesome!

Merci!
Lukas

cyberluke, 2022-08-22:
Also I will need to uncomment Custom FFB function here \driver\sys\hidReportDescSingle.h
`//0x09, HID_USAGE_CUSTM,   //    Usage ET Custom Force Data`

njz3, 2022-09-04:
Hi,
No the registry is not stored in the driver or in the source code anymore, and I removed the keys from the install file.
The reason is that windows 10 automatically adds the registry keys given the HID descriptor.
You can definitly overwrite the defaults keys by editing the registry.
Have you been able to compile and run with test mode (no signing required) on windows 11 ?

njz3, 2022-09-04:
> Also I will need to uncomment Custom FFB function here \driver\sys\hidReportDescSingle.h `//0x09, HID_USAGE_CUSTM, // Usage ET Custom Force Data`

This should also be possible to remove direclty in the registry stored HID.
The HID descriptor is stored and edited by vJoy Config, so it is worth give a look to it.

cyberluke, 2022-09-04:
Ok, thanks, I have seen it in vJoy Config.

I compiled it in Visual Studio 2015 on Windows 10. Tried VS2022 and 2019, but the instructions in Readme.md are no longer valid due to VS2019 being probably incompatible with that WDK+SDK.

I am Arduino hardware developer, so I run Windows 10 test mode by default (on top of that I use unsigned Radeon driver to output to VGA 15KHz Sony PVM professional CRT monitor).

VS2022 compiles it ONLY with the latest Windows 11 WDK + SDK also you need to download like 30 GB of files

After 5 days of daily work till morning hours, I solved it first in VS2022, which lead me to a solution for others.

Now I have a working VS2015 with WDK + SDK version 10.5xxx

Extra step I have to solve: The current project and WDK and SDK is not compatible with Windows 7 anymore. I had to manually edit all files and change "Windows7" string to "Windows10" string.

Next there was additional fatal error, I had to comment this part from official Windows SDK
C:\Program Files (x86)\Windows Kits\10\build\WindowsDriver.Common.targets
```
<!-- **
        Makes sure lower nt_target versions cant be linked to higher targert version libs.
       ** 
  <Target Name="ValidateNTTargetVersion" BeforeTargets="ValidateDriverProperties">
    <ValidateNTTargetVersion ValidNTTargetVersions   = "$(Valid_NTTARGETVERSIONS)"
                                             CurrentNTTargetVersion = "$(_NT_TARGET_VERSION)">
    <Output TaskParameter="NTTargetVersionNotValid" PropertyName="UnknownNTTARGETVERSION" />
    </ValidateNTTargetVersion>
  </Target>
 -->
```

Then I had to manually choose Windows platform for each project (SDK) and it must be exactly the same as WDK version installed (that one is not mentioned anywhere, there is only WindowsKernelDriver10.0 in Visual Studio and it is difficult to remember).

For me, there is no reason or advantage to update VJoy to latest Visual Studio as it only cause defects due to Microsoft itself. Right now they actively removed several stuff from Microsoft Downloads page, so I had to source a lot of stuff from Archive.org and I could not get exactly the same WDK + SDK that would allow me to build for Windows 7 and Windows 10.

My plan is to build for:
Windows XP
Windows 7
Windows 8.1
Windows 10

It is important for retrogaming community and also this awesome software did it out of the box before. So my goal is to update it and provide different individual project files or solution files for different Visual Studio and different platform, so it will work out of the box as it did with original repository. Sometimes it is also important to preserve functionality because sometimes it might have big value.

cyberluke, 2022-09-04:
About the registry, I found some strange issue:

VJoy reg file
```
@="GUID_ConstantForce"
"Attributes"=hex:26,00,0f,00,01,86,00,00,fd,03,00,00,fd,03,00,00,20,00,00,00
```

All my devices have it different (Thrustmaster Steering Wheel, Logitech Steering wheel, Logitech Force 3d and Logitech G940) have this:
```
@="Constant"
"Attributes"=hex:00,00,00,00,01,86,00,00,ed,03,00,00,ed,03,00,00,30,00,00,00
```

I am not able to decode all the values (would be really helpful, so I can translate the devices correctly)

Issues I encountered and solved:
1) The first number 26 vs 00 - that seems to be unique ID of the effect tied to the individual driver. It must be somewhere. I have also original Microsoft Force Feedback USB joystick - the VJoy PID is based on that, or was. This issue meant that I did not receive any FFB messages when importing the registry file of VJoy.

2) Then I tried to understand and match the attributes of my real device to VJoy for 1:1 mapping. I discovered that 3rd byte in VJoy attributes: 0f is required. When changed to 00, it will throw error invalid parameters. Then there is differenct bytes fd and ed. This causes some different number of axes or inverted axes behavior.

So my current solution, as I am rerouting all packets from VJoy to DirectX - SlimDX library, is that I provide a and recalculate mapping and values from VJoy to SlimDX. After 3 weeks of blood & sweat and working till 3am, I got a working solution. 

Only polar coordinates are not perfect yet because flightstick seems to have axis Y inverted.

I tried to work out the differences using VJoy code and/or registry of VJoy, but it did not help. Now I'm processing each packet and remapping axes. 

Also it still seems VJoy sends global gain 0%, even after my fix. I need to investigate that. Previous VJoy driver did not do it.

cyberluke, 2022-09-04:
Ad. self signed certificate: VS2015 can right click project and click on Driver -> Signing and in dropdown menu you can select Create self-signed certificate. VS2022 does not have this at all. For the best compatiblity, I choose VS2015 at this time.

cyberluke, 2022-09-04:
I also look at USB descriptor of real flightstick and it is completely different than the one VJoy is using. I believe VJoy is based on Microsoft Force Feedback. I have it laying right there, it might be better to debug it now with that one to get 1:1 experience, good catch!

Logitech USB descriptor contains only page 3 with vendor specific data. So my current approach was to write some bridge in FreePIE that translates VJoy to DirectX device and user can adjust the mapping in FreePIE GUI editor (simple Python script)

cyberluke, 2022-09-05:
Ok, verified with real Microsoft Force Feedback 2 usb joystick. Now it seems to behave ok. Will need to check some more details tomorrow. 

By the way I sent some pull request to your VJoy fork.

My next steps:
1) tweak VJoyConf utility (add custom force) and verify rudder/throttle corresponds now to flight stick Microsoft Force Feedback 2 default mapping
2) compile for Windows XP and get rid of unfinished makeovers from Steam (verified on dual boot PC that WinXP works much better for the era of force feedback flight simulators)