njz3/vJoyIOFeederWithFFB #10: Question: Lamp outputs

baritonomarchetto, 2022-09-19:
Hello njz3,

I am trying to understand your solution to interface Model2/3 cabinets with a PC and compare it with Aganyte and Sailorsat solutions.

There's one thing that I do not completely understand: I don't understand lamps/bulbs out management. 

If I take a look in example at [daytona 2  twin schematics](url=https://hico-srv022.pixhotel.fr/sites/default/files/gamoovernet/20201029153442-Totaly%20FuRy-Schema3.png) I can see the eight Tx and eight Rx signals on the bottom right side of the scheme. These are connected to Arduino MEGA D22-29 and D30-37 in your project, respectively. 
I can also see START and VIEW lamp on CN3 (schematics). This suggests to me that those lamps are not handled via messages on Tx/Rx buses, but directly from the game board. 

What am I missing? Are those outputs handled directly by dedicated arduino MEGA outputs?

Thanks for the stimulating work you did so far and for the help :)

njz3, 2022-09-19:
Hi,
The lamps outputs (CN3 in your schematics) are handled directly by the model 3 stack, and not by the driveboard.
In the case of the BFF, this is not directly handled by the Mega2560. Those lamps outputs are handled by a ULN2003 (transistors) that is connected to A8-A15 pins of the Mega2560. The lamps should have permanent 5V and the ULN2003 is making the GND connection when required to light the lamps. 
Note: there may be a mistake in your schematics as there should be 4x view buttons in daytona 2 twin cabinet?

baritonomarchetto, 2022-09-19:
Yes, you are right. By looking at [this fourth page of the schematics](https://hico-srv022.pixhotel.fr/sites/default/files/gamoovernet/20201029153454-Totaly%20FuRy-Schema4.png), four lamps are reported (one for view).

Yes, I am well aware that MEGA (or any other arduino board) cannot drive lamps and needs a driver IC like ULN2003/ULN2803. Thanks anyway (better safe than sorry!!).

I could not find the A8-A15 indication in the readme of this repository: is there an updated resource or link I should refer?

I like very much the vJoy+MEGA approach, I must say, and I think that a dedicated printed circuit board would be a step further in the spreading of this approach. Is there anything like that already?

njz3, 2022-09-19:
For the wiring, please have a look to the doc/ directory.
Those two files show how to wires the Mega2560 and the ULN2003
https://github.com/njz3/vJoyIOFeederWithFFB/blob/master/docs/schematics/Mega2560%20example.jpg
https://github.com/njz3/vJoyIOFeederWithFFB/blob/master/docs/schematics/Mega2560%20example%20sch.jpg

baritonomarchetto, 2022-09-19:
I see, thanks!

What about a dedicted PCB? Has it been done already? Something you can directly connect cabinets connectors without the need to cut wires?
I know that those cabs have some main differencies one with respect to the other, but nothing impedes the placement of more connectors on the same PCB...

njz3, 2022-09-19:
I personnaly only use dupont wires that I keep together with hot glue. This avoid to destroy the original loom and is still flexible to adapt to each cabinet.
Aganyte designed multiple hack boards to solder wires from a model 2 or 3 cabinet to a mega2560, but this does not follow my standard wiring (he has his own routing). In this case you have to tick the "alternative pinout for ffbcontroller" in my software.
Anyway, you are welcome to build your own hack board.

closed 2022-09-21 by baritonomarchetto