njz3/vJoyIOFeederWithFFB #11: Request: Support for custom pinout

baritonomarchetto, 2022-09-25:
Hello Benjamin, 
in the last days I have been working on a PCB to make wiring a genuine SEGA Model cabinet with PC simpler (in my hopes).
I adopted a slightly different arduino pinout from yours (digital inputs on pins 38-53 and outputs on A8-A15). I had no issues porting your code by modifying a single dependency of the main arduino code, but it's not generally a good idea to keep a fork separate from the seed.
I saw that the code already supports another pinout (Aganyte project, IIRC). Would it be possible to include also this into your BFF project?
The PCB will be the subject of an instructables and gerbers shared (this is not a product, but a labour of love, like yours BFF)

njz3, 2022-09-26:
Hi,
Yes I can add a listbox to select the pinout to use, then change the arduino code to use several alternative pin mappings.

baritonomarchetto, 2022-09-26:
That would be great. If you need more details I am at your disposal

baritonomarchetto, 2022-09-29:
Perfect, thanks!