njz3/vJoyIOFeederWithFFB #12: BFF crashes

baritonomarchetto, 2022-09-27:
I have already installed BFF on my win10 laptop for the very first tests with mame and a CH340 MEGA. That far, all good.
In order to test model 2 and supermodel I have set up a desktop PC with a dedicated videocard.
The new PC showed some difficulty with the CH340 MEGA clone I was using, suggesting some serial communication issue I have not resolved yet. I have then switched to a 16u2 MEGA 2560. This compiles ok, but when I launch BFF GUI, it crashes after recognizing the board. 
Is there a place where I can get a log file or similar in order to see what happens?
Notice that the new MEGA works in combination with BFF GUI on another PC (my laptop). No crashes at all there.

njz3, 2022-09-29:
If you run the software in console mode (BackForceFeeder.exe from command line) and not the GUI mode, if there is a crash it will give you an exception message that will be printed to the console.

baritonomarchetto, 2022-09-29:
Ok, this is what happens:

```
C:\Program Files\BackForceFeeder\BackForceFeeder-v0.6.0.68>backforcefeeder
Starting console mode of BackForceFeeder v0.6.0.68 Made for Gamoover by njz3!
****************************************
*** PRESS ESCAPE THEN ENTER TO LEAVE ***
****************************************
IMPOR|    1.773715:[MAIN] Machine ID=C28E2582C498B5A3
IMPOR|    1.775236:[MAIN] Invalid license.dat found for user DESKTOP-G1DMDR7@mail.com
IMPOR|    1.775347:[MAIN] No valid license.dat found, please contact the author and give your email and your machine ID to register
INFOR|    1.775665:[MAIN] Running in trial mode
IMPOR|    1.779471:[MAIN] Running as standard user, trying realtime priority
IMPOR|    1.783964:[MAIN] Setting realtime priority failed, got: High
IMPOR|    1.974649:[MANAGER] Program configured for PWM_SHAKER
IMPOR|    3.230073:[MANAGER] Found io board on COM3 version=0.2.0.0 type=IO BOARD ON MEGA2560
IMPOR|    3.454254:[MANAGER] Initializing IO board
IMPOR|    3.454849:[USBSerial] Performing IO board initialization
IMPOR|    4.489779:[USBSerial] IO board initialization done
INFOR|    4.490311:[MANAGER] Configuring IO board for pwmmode=0
INFOR|    4.509096:[MANAGER] Configuring IO board for wheelmode=2
INFOR|    4.526871:[MANAGER] Configuring IO board for pedalmode=0
INFOR|    4.544824:[MANAGER] Configuring IO board for ffbcontrollermode=0

Unhandled Exception: INFOR|    4.684267:[MAMEOutputsAgent] Entering HiddenWindow Thread
System.NullReferenceException: Object reference not set to an instance of an object.
   at BackForceFeeder.Inputs.RawInput.?(Boolean )
   at BackForceFeeder.Inputs.InputsManager. (UInt64 )
   at BackForceFeeder.Inputs.InputsManager. (Int32 , Int32 )
   at BackForceFeeder.BackForceFeeder.BFFManager. ()
   at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state)
   at System.Threading.ThreadHelper.ThreadStart()

C:\Program Files\BackForceFeeder\BackForceFeeder-v0.6.0.68>
```

njz3, 2022-09-29:
There is a bug in the input management. I must reproduce to fix it.
Can you attach your configurations and control sets files (as a zip) ?
Do you use the default arduino code ?

baritonomarchetto, 2022-09-29:
This was not the default code, but the one with my custom inputs. I had the crash also with the stock sketch anyway, so I would bet it will return the same error.
To be 100% sure I will upload your sketch tomorrow, relaunch the feeder, cooy the error and upload the zip you asked.
Thankyou

njz3, 2022-10-02:
I have reproduce the crash.
It is due to missing control set in the ControlSets direcotry (you don't even have a default one), which leads the software to have a null reference.
I will fix that. In the meantime, you can check by copying a default control set from another working installation.

njz3, 2022-10-02:
Should be fixed in 0.6.0.70, please give it a try

baritonomarchetto, 2022-10-02:
It actually is!

Thanks :)

closed 2022-10-02 by baritonomarchetto