njz3/vJoyIOFeederWithFFB #13: PWM_SHAKER question

baritonomarchetto, 2022-10-04:
Hello Benjamin,

may I ask: which pin is PWM rumble signal sent when PWM_SHAKER is choosed: is it arduino pin 22 (driverboard first bit)?

Thanks

njz3, 2022-10-04:
Hi,
It is the first PWM analog output pin that is usually mapped to D9 on a Leonardo or a Mega2560.

njz3, 2022-10-04:
I have seen that you have a JVS project to replace the Sega IO board.
Do you think I can implement the JVS protocol with a USB to RS485 adapter directly plug to a PC to drive the Sega IO board and avoid an Arduino ? (this would be for a F355 twin cab where I only want to swap the stack).

baritonomarchetto, 2022-10-04:
That replacement I/O board only emulates a joystick, but has not the ability to receive game outputs. Makes interfacing a SEGA JVS cab easier (no wire cutting), but at some cost.
It's out of my capabilities to handle outputs (that's your filed man :) ) 
There's an onboard driver circuit (ULN2003), so you can code some light game (i.e. a continuos blink), but nothing more ATM.

closed 2022-10-04 by baritonomarchetto