njz3/vJoyIOFeederWithFFB #18: Control vJoyIOFeederWithFFB with own program on PC

DikinCz, 2023-07-13:
Hello,
is there any way how could I control vJoyIOFeederWithFFB for BeamNG with my own program?
I have a program on PC that gets data from Arduino(Serial communication) and I want to control BeamNG with that data and get data about force feedback. I used Pyvjoy but it doesn't support force feedback. So everything I need is to set the axis and button value from my program on PC. Language doesn't have to be Python.
Thanks for any tips how to do it :)

pdaderko, 2023-07-24:
I did something similar to interface original Hydro Thunder arcade controls with BackForceFeeder.  I just uploaded the very barebones Python script that I wrote to basically emulate a FeederIOBoard, translating between BackForceFeeder and Python.  I put it in my fork of this project here: https://github.com/pdaderko/vJoyIOFeederWithFFB/blob/master/bff_translate/bff_translate.py .

I'll be uploading the Hydro Thunder specific code once I wrap up a few minor tweaks, though I don't think that'd be much more helpful than the above script for what you're doing.  I hope this helps!

njz3, 2023-08-08:
I started to include a Lua interpreter within the BackForceFeederto perform more customized actions and retrieve internal values.
A TCP socket with text commands might be opened to perform a king of "shell" interface to the BackForceFeeder.
Would that fit your needs ?
Regarding the emulation of an IO Board, I have never looked into as your are the first person to ask for.

DikinCz, 2023-08-08:
Hello, to be honest, I'm not best in Programming, I study electrotechnics. I have functional program, that reads data from Arduino and sends it to PC through serial communication and I need to set axis, buttons and send back data about FFB. Eventually I can buy Arduino Leonardo and control BeamNG directly. Actually, best for me, is to use commands like vJoy.AxisX = value. Unlikely I didn't find a lot of good tutorials or documentations, how to create own program for this. Thank you for help.

njz3, 2023-08-09:
@DikinCz : I am not sure I understand what you want to do.
If you already have an arduino that reads analog (wheel/pedals) and digital values (buttons) that you want to use to control BeamNG through a virtual joystick and get force feedback values, it is exactly what the BackForceFeeder does.

I would suggest you start from my Arduino code that integrates the communication procotol with the BackForceFeeder and modify the pins to read your own values. The forcefeedback values are also available within the Arduino code and you can do whatever you want with that : send it to another serial device, convert it to analog output, etc.