njz3/vJoyIOFeederWithFFB #19: FFB not working every time

aronkahrs-us, 2024-11-01:
I’ve been having an issue, mostly in Assetto Corsa, where I need to start a session multiple times before the force feedback starts working properly. It seems like the effects might be resetting after they’re set. I'm not using any effect tuning.
```
13:15:35 | INFOR| 9468,372787:[FFBMANAGER] [DEVICE_RESET] step 0	to [DEVICE_READY] step 0
13:15:35 | DEBUG| 9468,386697:[FFBMANAGER] Effect 2 [NO_EFFECT] to [DAMPER]
13:15:35 | DEBUG| 9468,386741:[FFBMANAGER] FFB Effect 2 got start effect
13:15:35 | INFOR| 9468,386748:[FFBMANAGER] [DEVICE_READY] step 0	to [DEVICE_EFFECT_RUNNING] step 0
```

njz3, 2024-11-01:
I don't see any particular issue in the extract of your log.
What I see is:
- A "Device Enable" command is received,
- A Damper effect is requested
- the effect runs
Do you mean the Damper is not active ?

Would be nice to give more detailled logs by enabling the following flag in the _ApplicationCfg.json_ file
[image]

aronkahrs-us, 2024-11-01:
I think what's happening is that after the effects are set, the system sends an 'effect running off' signal or something similar. This results in the effects appearing set, but with no force feedback while driving.

Below is a log from a session where FFB worked:
```
17:37:53 | DEBUG|   25,187105:[FFBMANAGER] FFB Got device reset
17:37:53 | INFOR|   25,187125:[FFBMANAGER] [DEVICE_READY] step 0	to [DEVICE_RESET] step 0
17:37:53 | DEBUG|   25,188778:[FFBMANAGER] FFB Got stop all effects
17:37:53 | DEBUG|   25,189068:[FFBMANAGER] FFB set device gain 1
17:37:53 | DEBUG|   25,189349:[FFBMANAGER] FFB Reset all effects
17:37:53 | INFOR|   25,189365:[FFBMANAGER] [DEVICE_RESET] step 0	to [DEVICE_READY] step 0
17:37:53 | DEBUG|   25,201245:[FFBMANAGER] Effect 1 [NO_EFFECT] to [CONSTANT_TORQUE]
17:37:53 | DEBUG|   25,202039:[FFBMANAGER] FFB Effect 1 got start effect
17:37:53 | INFOR|   25,202046:[FFBMANAGER] [DEVICE_READY] step 0	to [DEVICE_EFFECT_RUNNING] step 0
17:37:53 | DEBUG|   25,210434:[FFBMANAGER] Effect 2 [NO_EFFECT] to [DAMPER]
17:37:53 | DEBUG|   25,210477:[FFBMANAGER] FFB Effect 2 got start effect
```
And here’s a log from a session where FFB didn’t work:
```
17:40:02 | DEBUG|  154,208893:[FFBMANAGER] FFB Got device reset
17:40:02 | INFOR|  154,208913:[FFBMANAGER] [DEVICE_READY] step 0	to [DEVICE_RESET] step 0
17:40:02 | DEBUG|  154,208953:[FFBMANAGER] FFB Got stop all effects
17:40:02 | DEBUG|  154,208994:[FFBMANAGER] FFB set device gain 1
17:40:02 | INFOR|  154,213910:[FFBMANAGER] Device not yet ready
17:40:02 | DEBUG|  154,213962:[FFBMANAGER] FFB Effect 1 got start effect
17:40:02 | DEBUG|  154,213963:[FFBMANAGER] Device not yet ready
17:40:02 | DEBUG|  154,222821:[FFBMANAGER] FFB Reset all effects
17:40:02 | INFOR|  154,222837:[FFBMANAGER] [DEVICE_RESET] step 0	to [DEVICE_READY] step 0
17:40:02 | DEBUG|  154,229914:[FFBMANAGER] Effect 2 [NO_EFFECT] to [DAMPER]
17:40:02 | DEBUG|  154,229944:[FFBMANAGER] FFB Effect 2 got start effect
17:40:02 | INFOR|  154,229947:[FFBMANAGER] [DEVICE_READY] step 0	to [DEVICE_EFFECT_RUNNING] step 0
```
In the second log, despite the device eventually reaching the 'DEVICE_EFFECT_RUNNING' state, the FFB isn’t working.

I have enabled "VerbosevJoyFFBReceiver" but i don't know if i see any differences in the log.